qmk/qmk_configurator #564: Preonic rev2 loses layer 5 ("adjust" layer)

jdknezek, 2019-11-05:
After using the configurator to compile and download a firmware based on the `preonic/rev2` template, I lose access to layer 5 (the "adjust" layer), usually accessible by holding the raise & lower keys simultaneously.

In case it helps, my HID device ID is: `FEED:6061:0002` and my DFU device ID is `03EB:2FF4:0000`. I'm using qmk_toolbox 0.0.14.0 on Windows 10...

yanfali, 2019-11-05:
@jdknezek hey sorry,. But adjust isn't supported using lower and raise. Unfortunately it's implemented using custom code right now and isn't trivial to implement as a key code. We've discussed some work around privately but they will probably have to wait until the v2 API is finished. In the meantime you can simulate it by adding an adjust layer to one of your other layers and just holding a different key to get there

noroadsleft, 2019-11-05:
I've been meaning to fix that for a few months. :disappointed:

closed 2020-03-03 by yanfali