qmk/qmk_configurator #575: Configurator does not handle MT() correctly

SarreqTeryx, 2019-11-11:
If I hand code in "MT(KC_LSFT, KC_CAPS)", then try to load it into the Configurator, it reparses to "ANY(MT(KC_LSFT, KC_CAPS))"

I'm later realizing "RSFT_T(KC_CAPS)" exists, but doing it with MT() should at least reparse to that, rather than ANY().

noroadsleft, 2019-11-11:
It's because Configurator doesn't have any `MT()` keycodes directly in the UI. The `ANY()` wrapper allows users to use keycodes that the UI doesn't recognize. (The UI has no system in place for when a keycode requires multiple arguments, as `MT()` does.) The wrapper gets stripped from the data before Configurator sends it to the compiler, so it still works. I don't know that there's much demand for this at the moment.

yanfali, 2019-11-12:
@SarreqTeryx we've deliberately chosen not to support MT in the UI because we have the aliases. We could indeed reparse your MT code, and that might be an interesting thing to do. ANY isn't really designed for this use case though, so I'm going to close this as won't fix for now. Thanks for the input.

closed 2019-11-12 by yanfali