qmk/qmk_configurator #609: How to create and verify the info.json for the configurator?

yulei, 2019-11-30:
Currently the keyboard(https://github.com/qmk/qmk_firmware/tree/master/keyboards/matrix/noah) was rendered as all 1u key on the config.qmk.fm. While I use the same info.json on config.qmk.fm through the preview mode, it displayed correct( the number of keys was wrong).
 
[image]

I checked the layout fetching API, and found that the response returned with all "w" set to 1.
[image]

So it there still errors on my info.json file or I have missed anything?

noroadsleft, 2019-11-30:
I actually discovered this same issue last night.

Your contents are correct but the file encoding is wrong. The API expects the files it consumes to be encoded as UTF-8, but your `info.json` file is UTF-8 BOM. Unfortunately this is something that's hard to check via GitHub, and in fact I don't know any online resource that can check that. (I have my editor configured to use UTF-8 by default.)

yulei, 2019-12-01:
I'm using VS code or sublime, not sure while it was BOM coded :(   It will be great if you could help to fix this issue. thank you very much.

closed 2019-12-01 by yulei