qmk/qmk_configurator #623: strange qEvt() appears on the source code

yulei, 2019-12-13:
while I try to compile the [noah](https://config.qmk.fm/#/matrix/noah/LAYOUT_wkl) keyboard, I got the following error:
[image]

so I downloaded the source code and found 
[image]

the macro qEvt() was not from the chibios os, it was auto-generated and then inserted to the source codes? there should be some bugs which caused the macro inserted into line 934 lost one parameter.  Maybe a counter overflow? I have attached the file which caused the trouble.
[hal_i2c_lld.c.txt](https://github.com/qmk/qmk_configurator/files/3958939/hal_i2c_lld.c.txt)

yanfali, 2020-03-03:
@yulei was this fixed?

noroadsleft, 2020-03-04:
Looks fixed.

[image]

closed 2020-03-10 by yanfali