qmk/qmk_configurator #630: [Bug] Error compiling ergodox_infinity with ANY(KC__VOLUP)

focusaurus, 2020-01-02:
## Describe the Bug

I have a working keymap from config.qmk.fm for my ergodox_infinity that I made on 2019-01-28. I can compile that successfully with the latest qmk repo (via docker image). I went to tweak 2 key mappings via the web app and the new JSON file I downloaded won't compile.

Since it was such a surgical change, I pretty-printed and diffed both versions of the JSON files and discovered the problem was the change from `KC__VOLUP` in the older file vs `ANY(KC__VOLUP)` in the newer file. I changed just that line to `KC__VOLUP` and was able to compile the mapping successfully. So my assumption is something changed in the web app during that time. 

```
3c3
<   "keymap": "2019-01-28-qwe-navpunc-num",
---
>   "keymap": "2020-01-01-infinity-qwe-navpunc-num",
68c68
<       "KC__VOLUP",
---
>       "ANY(KC__VOLUP)",
104c104
<       "KC_TRNS",
---
>       "KC_PGUP",
130c130
<       "KC_TRNS",
---
>       "KC_PGDN",
```

My intended changes were just the `KC_PGDN` and `KC_PGUP` ones, the `KC__VOLUP` I didn't intentionally change I think it just changed due to the web app code being different.

## System Information

 - Keyboard:  ergodox_infinity
   - Revision (if applicable):  
 - Operating system: linux
 - AVR GCC version: 4.9.2 (I'm using the docker build)
 - ARM GCC version: 6.3.1 20170620 (release)
 - QMK Firmware version: 0.7.106

## Additional Context

Attaching the full error message from gcc as a text file.
[error.txt](https://github.com/qmk/qmk_firmware/files/4014850/error.txt)

noroadsleft, 2020-01-02:
Transferred because this seems to be a Configurator issue and not a Firmware issue.

---

This was probably caused by #562. We had a number of users who were confused by there being two sets of Volume keycodes, and using the wrong set and then reporting their Volume keys didn't work.

As the Windows/Linux volume codes also work on macOS, we removed the other set.

Also, I'm surprised you were able to get a working keymap from Configurator for an Ergodox Infinity; to my knowledge Configurator's never been able to compile firmware for it, unless I'm misunderstanding how you got your "working keymap."

focusaurus, 2020-01-02:
Hmm, I'm not sure about your last paragraph. I just went to the website, selected "ergodox_infinity", choose my keys and downloaded a JSON file. Then I was able to get that to compile to a `.bin`, so it seems supported throughout the stack.

focusaurus, 2020-01-02:
Oh let me be clear, I'm building the .json keymap using configurator then compiling that locally using the qmk_firmware docker image. I'm not trying to compile via the configurator web app.

noroadsleft, 2020-01-02:
Okay, I understand now.

There's two issues you ran afoul of. The first being the removal of the macOS volume keycodes as I described above. The second is that the `ANY()` wrapper is only designed for use within the QMK Configurator app itself (it interfaces with Configurator's UI code), and not within any other QMK tooling. Your removal of the wrapper to compile your firmware would be the correct solution here.

closed 2020-03-10 by yanfali