qmk/qmk_configurator #641: Not usable from mobile browser

KAndrsn, 2020-01-19:
The left side is cropped on Android Chrome and Android Firefox, making it impossible to look at any layer other than default.  I'm not planning to edit a layout, but I can't browse layouts when away from my desktop.

yanfali, 2020-05-18:
I think this would be a great idea for a mobile specific app. Since it doesn't need the full engine. It could just use the API data.

KAndrsn, 2020-05-20:
I was thinking the html/css could be adjusted so these parts are visible.  Then zoom and scrolling would be enough.

yanfali, 2020-05-20:
That's a pretty heavy weight solution to something that only needs the
layouts. I think if we were going to support mobile, it would probably be
best to actually do it properly. That's not the intent of this web
application.