qmk/qmk_configurator #647: RGB_MODE+/- not work for Redox/rev1

umezo, 2020-01-31:
When I put RGB_MODE +,- and some other LED control key codes, toggle, hue, sat and bright works but others doesn't work.

The binary of default layout downloaded from QMK Toolbox works well but compiling by qmk configurator dosen't.
(maybe this https://github.com/qmk/qmk.fm/blob/8c53fa0869675c9893f046c55bb226cdf4d5a048/compiled/redox_rev1_default.hex)

This is my json file for QMK configurator.
https://gist.githubusercontent.com/umezo/800f165bb569a9b294175703c69b9d5c/raw/23e9ccf53ee4403d294ba0593cf4da6f94e63aa1/umezo-keymap-redox.json

Should I build my self or is there some differences between each build of toolbox and configurator?

noroadsleft, 2020-01-31:
Wait, the `RGB Mode +` key works on the default keymap when downloaded through QMK Toolbox, but not on your keymap through Configurator? Do I understand that correctly?

umezo, 2020-02-04:
Sorry my describing.

I think "RGB_MOD" = "RGB_MODE +" and "RGB_RMOD" = "RGB_MODE -".

I downloaded binary of default keymap  from here.
[image]

After I flashed it, I could control RGB mode by following key.
[image]

But when I compiled my json and flashed its binary, this key doesn't work.

[image]

Actually, there are some keys work or don't work.

[image]

At last I tried compiling default keymap by GMK Configurator https://config.qmk.fm/#/redox/rev1/LAYOUT/.

It dosen't work.

Thanks!

yanfali, 2020-02-12:
You have to ask the maintainer of the keyboard. They put the RGB configuration in the default keymap not the keyboard level. That's why it doesn't work on configurator. Sorry.

closed 2020-02-12 by yanfali

noroadsleft, 2020-02-12:
> They put the RGB configuration in the default keymap not the keyboard level. That's why it doesn't work on configurator. Sorry.

That's what I missed.

https://github.com/qmk/qmk_firmware/blob/9456832a3bba4dd766a26a6286e78d3e88e4efce/keyboards/redox/keymaps/default/config.h#L29-L34

That should all be configured at the keyboard level, especially the `RGBLIGHT_ANIMATIONS` line.

kevprice83, 2023-04-17:
@umezo how did you solve this? I have the exact same issue and just sent a message to the maintainer of the keyboard but I suspect it's something we are able to configure and compile ourselves