qmk/qmk_configurator #655: Bug/Error when compiling new keymap for Matrix Noah

agbubby12, 2020-02-12:
As the title says, I am attempting to flash a new Matrix Noah and seem to be getting an error everytime I try to compile the keymap I want. Not sure what is going on because this is like my 2nd time using QMK, but I can attach the entire code I get here: 

[noah eror.txt](https://github.com/qmk/qmk_configurator/files/4189297/noah.eror.txt)

For context I am selecting matrix/noah as the keyboard and using the LAYOUT_default. Please let me know what other information I can give.

yanfali, 2020-02-12:
I've tested the docker container and it doesn't fail, so I'm not sure what's going here. It could be something to do with the version of the source pulled down by the API being out of date.

yanfali, 2020-02-12:
Looks like this fix is needed: https://github.com/qmk/qmk_compiler/pull/59

skullydazed, 2020-02-12:
This should be working now that I've pushed out https://github.com/qmk/qmk_compiler/pull/59. I was able to compile an empty layout just now.

closed 2020-02-12 by skullydazed