qmk/qmk_configurator #668: Whole body moves while dragging (& can't push to new branch)

robinrabauke, 2020-03-02:
Hi,

I've realised that the whole body moves slightly to the right while I drag a key from the keyboard to the keymap. This also makes it hard to actually drag a key.

I've reintroduce the class `.hidden` to actually hide the drag-container, which works but I can't push to the repository.

What do I need to do to be able to create a branch and a PR?

best, robin

yanfali, 2020-03-03:
yeah, PR. https://opensource.com/article/19/7/create-pull-request-github
What projects let you push code to them without permissions?

yanfali, 2020-03-03:
@robinrabauke please close issue after you've verified it's fixed for you

robinrabauke, 2020-03-03:
> @robinrabauke please close issue after you've verified it's fixed for you

I was not aware that I need to fork it. Makes total sense now :) Thanks for the clarification!

closed 2020-03-03 by robinrabauke