qmk/qmk_configurator #670: QMK Configurator doesn't work for Ergodox Infinity

NFiorentini, 2020-03-03:
`QMK Firmware 0.8.13`
`WARNING: Some git submodules are out of date or modified.`
 `Please consider running make git-submodule.`

`Making ergodox_infinity with keymap ergodox_infinity_layout_ergodox_help`

`tmk_core/chibios.mk:38: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x7.mk: No such file or directory`
`tmk_core/chibios.mk:40: lib/chibios/os/hal/hal.mk: No such file or directory`
`tmk_core/chibios.mk:50: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: No such file or directory`
`tmk_core/chibios.mk:82: lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1/board.mk: No such file or directory`
`tmk_core/chibios.mk:86: lib/chibios/os/rt/rt.mk: No such file or directory`
`tmk_core/chibios.mk:92: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: No such file or directory`
`tmk_core/chibios.mk:94: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory`
`quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: No such file or directory
makek'.  Stop.`
`Make finished with errors`
`Makefile:560: recipe for target 'ergodox_infinity:ergodox_infinity_layout_ergodox_help' failed
make: *** [ergodox_infinity:ergodox_infinity_layout_ergodox_help] Error 1`

yanfali, 2020-03-03:
Sorry, this is a licensing issue. qmk firmware is working on removing this dependency. the licenses are incompatible with distribution.

yanfali, 2020-03-03:
Dupe https://github.com/qmk/qmk_configurator/issues/201

closed 2020-03-03 by yanfali

NFiorentini, 2020-03-03:
Does this mean that QMK firmware will never support the Ergodox Infinity?

yanfali, 2020-03-03:
It means you have to use the source code and build it from the cli. Sorry.
It's a licensing issue, we aren't allowed to legally distribute binaries.
Thank Input.Club.

skullydazed, 2020-03-03:
To expand a bit on this issue-

When the ergodox infinity was first implemented in QMK we were not very good about checking licenses. It turns out that [ugfx](https://ugfx.io/) is not GPL compatible, and we can't distribute binaries that comply with both that license and the GPL.

What is needed is for someone to write an alternative LCD implementation for the infinity, something we want to do. Once that happens it will be supported in configurator.

NFiorentini, 2020-03-03:
Thank you for the explanation.  Currently, the kiibohd Configurator, current and previous versions, does not handle layer switching properly, and the issue has been outstanding since last July.  I had brought the issue up to Haata in November, and he isn't responding now.  I can't get anyone on /r/MechanicalKeyboards to help.  Is there anyone else that I talk to or anything that I can do?

NFiorentini, 2020-03-04:
I've contacted the ugfx licence owners listed [here](https://ugfx.io/license.html), asking for their help in remedying this situation.  I just want to get this keyboard working again.

skullydazed, 2020-03-04:
Even if we have permission (they have actually given it before) it's not a situation we're comfortable with and will not be enabling ugfx in configurator.

NFiorentini, 2020-03-04:
Wish we had private messaging here, because I'd really like to know more.

What else can I do to get this keyboard working again?  I'm willing to pay someone to teach me.

skullydazed, 2020-03-04:
We use [discord](https://discord.gg/Uq7gcHh) for realtime chat. It's often a great way to have back and forth conversations about these topics.

Your quickest way to get it working again is to use the source and compile it locally. We have instructions for that here: https://docs.qmk.fm/#/newbs

NFiorentini, 2020-03-05:
I received the following response from uGFX:

> Having read the transcript of your issue, the problem with the compile is with ChibiOS – not uGFX. It looks like there are missing components in ChibiOS that your makefile is expecting.

>You will need to contact ChibiOS in relation to the build problems you are experiencing there.

>uGFX can be distributed in binary form PROVIDED a commercial license has been purchased. We even provide free commercial licenses for certain types of projects e.g. open-hardware projects.

>The issue that they are mentioning with regard to binary distribution is likely because of ChibiOS’s GPL license. Subject to the above mention of having a commercial uGFX license, it is the GPL license that is preventing the distribution. We have no such exclusion in our license. 

>I believe that ChibiOS is also available in a non-GPL commercial license which contains no restrictions on binary distribution. Please contact ChibiOS in relation to that.

>We wrote a really good article on the interaction between GPL and any other sort of license such as the uGFX license, the BSD license, the Mozilla license etc but I can’t seem to find it.

>Perhaps Joel can find the article?

>Regards,

>Andrew.

Does this change things at all?

skullydazed, 2020-03-05:
That reinforces our point- it is not possible for us to comply with terms of the GPL and the terms of the uGFX license at the same time. That's why we made the deliberate choice to exclude uGFX from API builds.