qmk/qmk_configurator #672: Designer Consent for Colorways Updates

xplusplus, 2020-03-04:
Gathering consent from keyset designers for updates to the available colorways in the configurator. 

Currently I'm adding keyset designs found in the 2020 Mech Madness Design bracket. This issue will track the consent to add the colorways which I've gained from reaching out to the designers.

_Originally posted by @yanfali in https://github.com/qmk/qmk_configurator/pull/665#issuecomment-594108740_

@xplusplus

xplusplus, 2020-03-04:
- GMK 8008
    - Permission from Dixie granted
- GMK 9009 (R3)
    - Permission TBD
- GMK Alter
    - TBD
- GMK Analog Dreams
    - Permission from pwade3 granted
- GMK Ascii
    - Permission from Oblotzky granted
- GMK Bingsu
    - TBD

skullydazed, 2020-03-04:
Please track these permissions in the original PR so that everything is in one place.