qmk/qmk_configurator #677: keymap download broken

elmo-space, 2020-03-10:
Downloading the compiled keymap doesn't work anymore right now.
Instead of downloading a zip file with the keymap.c the "keymap only" button gives you the same file as the "keymap json" download button, just with a different filename.

skullydazed, 2020-03-10:
This is a new change. You can compile JSON files using the CLI and have been able to for some time. The API has been changed to use that facility instead of generating a keymap.c, so it's delivering the keymap source as used.

yanfali, 2020-03-10:
That means the UI is broken. Re-opening and renaming the bug

yanfali, 2020-03-10:
ok. discussed this offline. @skullydazed will be opening an issue in future to aggregate future changes for the UI. closing for now.

closed 2020-03-10 by yanfali