qmk/qmk_configurator #684: Compiling issues for helix\rev2

b1uegh0st, 2020-03-13:


b1uegh0st, 2020-03-13:
(sorry hitting enter too soon)
Since the earlier of this week, everytime I try to compile a helix/rev2 (any of them except for the split common) the compiler resulting in error, even with default keymap load by the tool. 

Picture attached. Please help.

[image]

mtei, 2020-03-13:
As a workaround, use `helix/rev2/sc/back` instead of `helix/rev2/back`.

This is an experimental implementation, but you can use it for now.

Look here if you want to know the difference between the two. (The difference is a little.) Https://github.com/qmk/qmk_firmware/pull/7718

mtei, 2020-03-13:
I'm sorry, but when I tried it, I realized that the back light couldn't be used.

b1uegh0st, 2020-03-13:
Thanks for your answer. I wasn't sure what SC really means (and if it is usuable) so did not try it while in a hurry. Will re-try this weekend, not caring about the backlight for now.

b1uegh0st, 2020-03-17:
So I tried a keymap today on helix/rev2/sc. Somehow one row is off (only map the left side and connected the left side).

json I used: https://www.dropbox.com/s/8pfqryw3fofhd73/helix_sc_left.json?dl=0
hex that is compiled: https://www.dropbox.com/s/kl9034s8wyoxl5s/helix_rev2_sc_helix_sc_left.hex?dl=0

b1uegh0st, 2020-03-17:
(deleted comment) My bad, mapped the wrong file.

mtei, 2020-12-11:
I made a workaround for the compile error on QMK configurator.

Currently the following two targets are compilable.
* helix/pico/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)
* helix/rev2/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)

b1uegh0st, 2020-12-12:
Hi @mtei , just an FYI, the helix/rev2/sc does not seem to have media key (vol+, play) working. I suspect it to not have "EXTRAKEY_ENABLE = yes"

mtei, 2020-12-14:
`helix/rev2/sc` is an experimental target and is not recommended for now. You should probably be able to use `helix/rev3_5rows` or` helix/rev3_4rows` instead. Because helix rev3 is upward compatible hardware of helix rev2. (However, helix rev3 has not been released yet, only the firmware is complete.)

b1uegh0st, 2020-12-28:
Noted with thanks. All good now.

closed 2021-05-20 by mtei