qmk/qmk_configurator #685: Feature request: Supporting separate maps for left/right of the split keyboard

b1uegh0st, 2020-03-13:
This maybe a bit much to ask (as it may lengthen the list of keyboard supported). Yet here is the use case:

- I do like to map my helix on QMK Configurator. 
- Recently decided to use the 2 halves separately (each with its own USB connection) as macro pads. 
- I can use QMK Configurator to map the left half ok.
- For the right half I need to reverse visualize where certain key will be and just map the left-side accordingly (which is treated as master, by default of config.h - which is not yet editable)

Hope this is clear enough.

yanfali, 2020-06-15:
it's not entirely clear what you want here. Can you post some examples? Are you talking about programming the left and right hands of the keyboard separately?

b1uegh0st, 2020-06-22:
Yes I am talking about programming them (as well as mapping them) separately. 

Currently if I only have the right-hand one, and plug that in, using QMK Configurator keymap wouldn't work (the key got flipped around).