qmk/qmk_configurator #702: The download button for the keymap.c downloads the json

sanjibukai, 2020-04-09:
Hi everybody,
Shouldn't the button in the far left **KEYMAP ONLY** download the `keymap.c` file?
Currently in download the JSON file.

noroadsleft, 2020-04-10:
Duplicate of #677 and #695.

This issue is part of the work happening under #692. Please see that issue for updates.

Closing.

closed 2020-04-10 by noroadsleft

noroadsleft, 2020-04-10:
@sanjibukai,

Apologies, forgot to actually give you a real reply.

Configurator's keymaps have long been natively JSON. Previously, Configurator's API would convert the JSON data into a `keymap.c` file, then compile that to provide the user's firmware.

Recently that API underwent a change where it now compiles directly from the JSON data, instead of converting the file.

If you need the `keymap.c` file, the most straightforward way to get one is to set up the [QMK CLI](https://docs.qmk.fm/#/cli), and run:

```
qmk json2c -o [path where you want to save keymap.c] [path to your exported keymap.json]
```

If you need any help let us know.

sanjibukai, 2020-04-10:
Hello,
I apology too since I didn't check for existence of similar issue..
I know that it's possible to compile from JSON, but I need the keymap.c file because I need to add customization like layer LED lighting (I have a basic ergoidox) as well as some custom sequence generation (using SEND_STRING).
Anyway thank for this great environment, I used QMK some years ago, and many things change meanwhile!

skullydazed, 2020-04-10:
>  I need the keymap.c file 

You can convert the JSON to a C yourself using `qmk json2c <file.json>`.