qmk/qmk_configurator #711: Getting error on compile in QMK Configurator: tmk_core/chibios.mk:123: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory

danmchan, 2020-04-19:
I've been using QMK Configurator for a while now, but just recently, it stopped working on my ProjectKeyboard/Alice layouts. 

The following is the JSON file used:
[malice (7).txt](https://github.com/qmk/qmk_configurator/files/4500361/malice.7.txt)

Full output below:
* Sending projectkb/alice:malice.json with LAYOUT_default
* Received job_id: fb5a3eb2-f9bd-4f82-a9fc-98bed93a21df
* Running . . . . .
* Finished:



Making 

tmk_core/chibios.mk:123: lib/chibios/os/hal/lib/streams/streams.mk: No such file or directory
makek'.  Stop.
make: *** alice.json] Error 1
ake finished with errors

noroadsleft, 2020-04-20:
Hello @danmchan,

QMK just in the last week gained support for the rev2 of Project Keyboard's Alice PCB, and the addition of the new revision necessitated relocating the files for compiling rev1 firmware.

You can fix this in the short term by opening your keymap JSON file in a text editor, and finding the part that says `"keyboard":"projectkb/alice"` and changing it to read `"keyboard":"projectkb/alice/rev1"`.

Expect the `projectkb/alice` option in QMK Configurator to disappear in the next day or two, replaced by options `projectkb/alice/rev1` and `projectkb/alice/rev2`.

danmchan, 2020-04-20:
Thanks James. Makes sense. I’ll try that out.

danmchan, 2020-05-02:
That worked by the way. Thanks for your help!

closed 2020-05-06 by noroadsleft