qmk/qmk_configurator #714: Build error when compiling any of the Helix/Rev2 keyboards

agaisin, 2020-04-20:
Errors here: **Linking: .build/helix_rev2_layout_mine.json.elf                                                     
 | 
 | .build/obj_helix_rev2_layout_mine.json/keyboards/helix/rev2/rev2.o: In function `matrix_init_kb':
 | /qmk_compiler/qmk_firmware/keyboards/helix/rev2/rev2.c:25: undefined reference to `is_helix_master'
 | collect2: error: ld returned 1 exit status
 | 
makeine.json.elf] Error 1
make: *** ine.json] Error 1
ake finished with errors**

Compiler output below.
-----------------------------
Helix
===

[image]

A compact split ortholinear keyboard.

Keyboard Maintainer: [Makoto Kurauchi](https://github.com/MakotoKurauchi/) [@pluis9](https://twitter.com/pluis9)  
Hardware Supported: Helix PCB Alpha, Beta, Pro Micro  
Hardware Availability: [PCB &amp; Case Data](https://github.com/MakotoKurauchi/helix), [Yushakobo Shop](https://yushakobo.jp/shop/), [Little Keyboards](https://littlekeyboards.com/collections/helix) 

## How to build
 * [Helix how to Customize and Compile](rev2/keymaps/default/readme.md#customize)
 * [HelixPico how to Customize and Compile](pico/keymaps/default/readme.md#customize)

See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.

Loaded 6 layers and 384 keycodes. Defined 4 Any key keycodes
* Sending helix/rev2:layout_mine.json with LAYOUT
* Received job_id: a93c5fea-4356-46c5-b6f4-e9d49f91c24f
* Running . . . .
* Finished:



Making 

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/helix/helix.c                                                                  
Compiling: keyboards/helix/rev2/rev2.c                                                              
Compiling: keyboards/helix/rev2/keymaps/layout_mine.json/keymap.c                                   
Compiling: quantum/quantum.c                                                                        
Compiling: quantum/keymap_common.c                                                                  
Compiling: quantum/keycode_config.c                                                                 
Compiling: quantum/matrix_common.c                                                                  
Compiling: quantum/matrix.c                                                                         
Compiling: quantum/debounce/sym_g.c                                                                 
Compiling: quantum/process_keycode/process_space_cadet.c                                            
Compiling: quantum/process_keycode/process_magic.c                                                  
Compiling: quantum/process_keycode/process_grave_esc.c                                              
Compiling: tmk_core/common/host.c                                                                   
Compiling: tmk_core/common/keyboard.c                                                               
Compiling: tmk_core/common/action.c                                                                 
Compiling: tmk_core/common/action_tapping.c                                                         
Compiling: tmk_core/common/action_macro.c                                                           
Compiling: tmk_core/common/action_layer.c                                                           
Compiling: tmk_core/common/action_util.c                                                            
Compiling: tmk_core/common/print.c                                                                  
Compiling: tmk_core/common/debug.c                                                                  
Compiling: tmk_core/common/util.c                                                                   
Compiling: tmk_core/common/eeconfig.c                                                               
Compiling: tmk_core/common/report.c                                                                 
Compiling: tmk_core/common/avr/suspend.c                                                            
Compiling: tmk_core/common/avr/timer.c                                                              
Compiling: tmk_core/common/avr/bootloader.c                                                         
Assembling: tmk_core/common/avr/xprintf.S                                                           
Compiling: tmk_core/common/magic.c                                                                  
Compiling: tmk_core/protocol/lufa/lufa.c                                                            
Compiling: tmk_core/protocol/usb_descriptor.c                                                       
Compiling: tmk_core/protocol/lufa/outputselect.c                                                    
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 
**Linking: .build/helix_rev2_layout_mine.json.elf                                                     
 | 
 | .build/obj_helix_rev2_layout_mine.json/keyboards/helix/rev2/rev2.o: In function `matrix_init_kb':
 | /qmk_compiler/qmk_firmware/keyboards/helix/rev2/rev2.c:25: undefined reference to `is_helix_master'
 | collect2: error: ld returned 1 exit status
 | 
makeine.json.elf] Error 1
make: *** ine.json] Error 1
ake finished with errors**


skullydazed, 2020-04-21:
The helix board requires special flags passed to the make command which the API can't support. I'm afraid someone will have to some work to make it work on configurator.

mtei, 2020-04-21:
> The helix board requires special flags passed to the make command which the API can't support. I'm afraid someone will have to some work to make it work on configurator.

I have published #8422. This is a change on the QMK CORE side to resolve this issue. (I think so.)

And I think the change on the Helix keyboard side would be solved by adding two small files after merging #8669.

mtei, 2020-12-11:
I made a workaround for the compile error on QMK configurator.

Currently the following two targets are compilable.
* helix/pico/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)
* helix/rev2/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)

fauxpark, 2022-12-13:
I believe this is fixed now.