qmk/qmk_configurator #729: Compilation fails for /hhkb/LAYOUT_JP

CommandLineDesign, 2020-05-06:
It seems there are a lot of issues for HHKB-JP Layout. For one, the default keymap is blank. For two, it seems it's not possible to compile even a very simple layout for it.

Here's an example from a very simple test with no functions or layers:

```

HHKB Alternate Controller
===

An alternative controler for the HHKB designed by hasu.

Keyboard Maintainer: QMK Community  
Hardware Supported: HHKB Alternate Controller  
Hardware Availability: https://geekhack.org/index.php?topic=12047.0  

Make example for this keyboard (after setting up your build environment):

    make hhkb:default

See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.
* Sending hhkb:hhkb-jp-test with LAYOUT_JP
* Received job_id: 03b6123a-5ddb-47ee-8d1c-c1f5c79beb32
* Running . . . . . . . .
* Finished:



Making 

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/hhkb/matrix.c                                                                  
Compiling: keyboards/hhkb/hhkb.c                                                                    
Compiling: keyboards/hhkb/keymaps/hhkb-jp-test/keymap.c                                            In file included from [K:
ents in array initializer [-Werror]
     { KC_NO, KC_NO, K82,   K83,   K84,   K85,   K86,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, K82,   K83,   K84,   K85,   K86,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, K92,   K93,   K94,   K95,   K96,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, K92,   K93,   K94,   K95,   K96,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KA2,   KA3,   KC_NO, KA5,   KA6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KA2,   KA3,   KC_NO, KA5,   KA6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KB2,   KB3,   KB4,   KB5,   KB6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KB2,   KB3,   KB4,   KB5,   KB6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KC2,   KC_NO, KC4,   KC5,   KC6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KC2,   KC_NO, KC4,   KC5,   KC6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KD2,   KD3,   KD4,   KD5,   KD6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KD2,   KD3,   KD4,   KD5,   KD6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KE2,   KE3,   KE4,   KE5,   KE6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KE2,   KE3,   KE4,   KE5,   KE6,   KC_NO },                \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
ents in array initializer [-Werror]
     { KC_NO, KC_NO, KF2,   KF3,   KF4,   KF5,   KF6,   KC_NO }                 \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
[K’)
     { KC_NO, KC_NO, KF2,   KF3,   KF4,   KF5,   KF6,   KC_NO }                 \
[K
[K’
  , KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_RSFT, KC_LCTL, KC_LCTL, KC_LGUI, KC_LALT, KC_LSFT, KC_SPC, KC_RSFT, KC_RALT, KC_RGUI, KC_RCTL, KC_LEFT, KC_DOWN, KC_RGHT)
[K
cc1: all warnings being treated as errors
 
 | 
 | 
 | 
makeap.o] Error 1
make: *** akefile:579: hhkb:hhkb-jp-test] Error 1
ake finished with errors

```

CommandLineDesign, 2020-05-06:
I was looking at recent changes to the HHKB keyboard section and noticed this: https://github.com/qmk/qmk_firmware/pull/8618/files

Is it necessary to add the `LAYOUT_JP` to the rules.mk in the same manner `LAYOUT_60_hhkb` was added?

CommandLineDesign, 2020-05-06:
After manually running qmk json2c with the exported json file, I am able to compile and flash the keyboard successfully, however the layout does not match the UI's config, there seems to be an incorrect layout definition for HHKB_JP

CommandLineDesign, 2020-05-06:
Was able to verify that the layout_jp entry in https://github.com/qmk/qmk_firmware/blob/master/keyboards/hhkb/info.json is incorrect. "Return" should be at `"y": 2` instead of` "y": 1` on line 103 `{ "label": "Enter", "x": 13.75, "y": 2, "w": 1.25, "h": 2 },`

Not sure if this change will negatively affect the UI of the QMK_Configurator tool though.

And also, still not clear why compilation was failing, I assume it's related to this problem?

CommandLineDesign, 2020-05-06:
I believe the issue is that the `LAYOUT_JP` entry in `/keyboards/hhkb/info.json` is incorrect.

The Enter key: is in `"y": 1` and it should be in `"y": 2`

https://github.com/qmk/qmk_firmware/blob/master/keyboards/hhkb/info.json#L103

noroadsleft, 2020-05-06:
The reason for this is the existing HHKB codebase requires a define to be set in order to compile firmware for the board, but Configurator's API doesn't have any support for changing build options or setting extra defines, so only the ANSI version can be compiled.

There's actually already a PR open that will allow the HHKB JP to be compiled, but it won't land on QMK until the end of the month.

CommandLineDesign, 2020-05-06:
@noroadsleft Great, thanks for your reply! 🙏

noroadsleft, 2020-06-17:
This is fixed btw. Sorry for the late notice.

https://config.qmk.fm/#/hhkb/jp/LAYOUT_JP

closed 2020-06-17 by noroadsleft