qmk/qmk_configurator #730: NoSuchKeyboardError

jackassplus, 2020-05-11:
# Clueboard 2x1800

Clueboard Double 1800 All The Way

* Keyboard Maintainer: [Zach White](https://github.com/skullydazed)
* Hardware Supported: Clueboard 2x1800 PCB
* Hardware Availability: 2018 Apr 1 Group Buy

Make example for this keyboard:

    make 2x1800:default

See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.

Loaded 1 layers and 127 keycodes. Defined 0 Any key keycodes
* Sending clueboard/2x1800:clueboard_2x1800_layout_mine with LAYOUT
* Received job_id: 63ff1165-0f7b-40c9-8e19-d9d7eb00bd6b
* Running . . .
* Finished:

<class 'qmk.errors.NoSuchKeyboardError'>
aps directory for: clueboard/2x1800
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/milc.py", line 608, in __call__
    return self.__call__()
  File "/usr/local/lib/python3.7/dist-packages/milc.py", line 613, in __call__
    return self._entrypoint(self)
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/decorators.py", line 40, in wrapper
    return func(*args, **kwargs)
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/decorators.py", line 83, in wrapper
    return func(*args, **kwargs)
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/cli/compile.py", line 35, in compile
    keymap_path = qmk.path.keymap(user_keymap['keyboard'])
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/path.py", line 57, in keymap
    raise NoSuchKeyboardError('Could not find keymaps directory for: %s' % keyboard)
qmk.errors.NoSuchKeyboardError: Could not find keymaps directory for: clueboard/2x1800

noroadsleft, 2020-05-11:
The 2x1800 got an update last week where it has two revisions, and you have to know which revision you have in order to compile.

If you have the Mechanical Drawing Toy Edition, use [clueboard/2x1800/2019](https://config.qmk.fm/#/clueboard/2x1800/2019/LAYOUT).

If you have the Double 1800 All The Way, use [clueboard/2x1800/2018](https://config.qmk.fm/#/clueboard/2x1800/2018/LAYOUT).

jackassplus, 2020-05-11:
got past that, running into different errors in the qet interface that I can't read, so I'll compile in Linux and see if I can figure it out.

closed 2020-05-11 by jackassplus