qmk/qmk_configurator #734: KBDFans Tofu96 missing

McCroden, 2020-05-19:
The [KBDFans Tofu96](https://kbdfans.com/collections/group-buy/products/tofu96-mechanical-keyboard-diy-kit) has shipped and is not available yet in qmk configurator. 

I asked KBDFans for a link to the firmware and was provided:
This url: http://www.qmkeyboard.cn/
Choose "DZ96" for the appropriate qmk firmware.

I'm new to QMK--apologies if this is not the correct avenue to get the board added to QMK configurator.

yanfali, 2020-05-19:
Yeah, sorry that's the wrong way. You file an issue on qmk_firmware, then it shows up in configurator. You can do most of the work yourself using the JSON file from the chinese web site using https://noroadsleft.github.io/kbf_qmk_converter/

closed 2020-05-19 by yanfali

noroadsleft, 2020-05-20:
@McCroden Based on the data in the DZ96 JSON file from qmkeyboard.cn, the [Espectro](http://localhost:8081/#/espectro/LAYOUT_default) may be the same PCB.

McCroden, 2020-05-20:
@noroadsleft thank you! I'll look into it.

McCroden, 2020-05-20:
In case others find this post, @noroadsleft recommendation does work. If there are 2 keys in the configurator where you have 1 key, then map both keys to what you want that 1 key to be.