qmk/qmk_configurator #741: Default Plaid layout fails on compile

benjaminmikiten, 2020-05-31:
Steps to reproduce:
1. Navigate to https://config.qmk.fm/#/dm9records/plaid/LAYOUT_plaid_grid
2. Using the default layout, press compile.

Result:
Compilation fails. Error given: 
`[Kusbdrv/usbdrv.h: No such file or directory ...`

I assume it's looking for `/tmk_core/protocol/vusb/vusb.h` and that `kusbdrv` is a typo, but I can't  find said typo anywhere.  

I was able to reproduce this with a local version of the configurator as well.

noroadsleft, 2020-06-04:
Sorry it took me so long to get back to you on this.

Right now boards that use V-USB aren't compiling in Configurator, and we haven't solved it yet.

As a work around, you can fork the qmk/qmk_firmware repository, set up the [QMK CLI](https://docs.qmk.fm/#/cli), and compile the keymap with that.

1. Export the keymap JSON from Configurator. Save it to the root qmk_firmware directory from your fork.
2. Compile the keymap using `qmk compile`:
   ```
   QMKuser /d/GitHub/qmk_firmware (master)
   $ qmk compile dm9records_plaid_layout_plaid_grid_mine.json
   Ψ Wrote keymap to /d/GitHub/qmk_firmware/keyboards/dm9records/plaid/keymaps/dm9records_plaid_layout_plaid_grid_mine/keymap.c
   Ψ Compiling keymap with make dm9records/plaid:dm9records_plaid_layout_plaid_grid_mine
   ...
   ```
3. The compile job will place your compiled firmware in the root qmk_firmware directory. Flash this using your normal flashing method.

benjaminmikiten, 2020-06-05:
Good to know, thank you so much! I'll close this since this workaround works and I assume that the larger vusb issue is being addressed somewhere else.

closed 2020-06-05 by benjaminmikiten