qmk/qmk_configurator #744: Compilation error on default layout for: ymdk_np21/LAYOUT_ortho_6x4

philipmgrant, 2020-06-07:
Compiling the default layout for ymdk_np21/LAYOUT_ortho_6x4 fails with:

```
error: pathspec 'lib/vusb' did not match any file(s) known to git
make: *** ine] Error 1
```

This seems to be fairly new: my own layout for this board compiled fine on QMK Configurator on 30th May but fails now.

noroadsleft, 2020-06-11:
There is currently a known issue with Configurator not being able to compile firmware for boards that use V-USB – namely, anything powered by an atmega32a, atmega328, atmega328p, or attiny85. The ymdk_np21 is powered by an atmega32a.

We are working on a fix, but it hasn't been deployed yet.

noroadsleft, 2020-06-11:
This is now fixed.

philipmgrant, 2020-06-11:
Works perfectly now - thanks very much!

closed 2020-06-11 by philipmgrant