qmk/qmk_configurator #751: Change Mouse_enabled to yes in Rules.mk for Keeb.io Quefrency boards

jdbaggins, 2020-06-18:
I'm not much of a developer and have a windows machine I'd prefer to keep clean, so I am largely unable to update the Rules.mk for the Quefrency board I just built, or I'm just unaware of how to make the appropriate changes in Configurator. 

I largely rely on my mouse setting for my other boards for navigating, so the option is sorely missed on my Quefrency.

Could you change the basic Quefrency rules.mk to set the Mousekey_enabled to true by default?

Also, I am curious why the setting is not enabled by default on all boards, are certain ones unable to do so due to hardware constraints?

noroadsleft, 2020-06-19:
I'm not sure why @nooges left Mouse keys disabled for the Quefrency, but I suspect it may be due to the firmware size available on the microcontroller.

I have to leave in a bit, but if he doesn't reply by the time I get back home I'll ask him about it.

nooges, 2020-06-19:
For rev1, there's enough space for mousekeys, but adding it in for rev2 would makes things a bit tight.

Okay for someone to add it to rev1/rules.mk.

noroadsleft, 2020-06-20:
> Okay for someone to add it to rev1/rules.mk.

Thanks, Danny. I'll open a PR for this.

@jdbaggins If you need firmware for a Quefrency rev2 with Mousekeys enabled, contact me on Discord at `xxiinophobia#5825` and I can see about getting you sorted.

noroadsleft, 2020-06-20:
@jdbaggins K, Mousekeys is enabled for Quefrency rev1 now.

jdbaggins, 2020-06-20:
Awesome, thanks everyone, I'll give it a shot.

jdbaggins, 2020-06-23:
Sorry for the delay, everything looks great! mousekeys work great. Thanks everyone.

closed 2020-06-23 by jdbaggins