qmk/qmk_configurator #752: Not Loading Custom QMK Keymap  JSON

otaviocc, 2020-06-18:
When I try to load a valid custom JSON, it gives me the follow screen, no matter if loading from the disk, or from the cloud.

[image]

I've tried Firefox, Firefox Private Mode, Google Chrome and Chrome Private Mode. Same result in all of them.

yanfali, 2020-06-18:
@noroadsleft my guess is the default still references a layout that no longer exists LAYOUT_preonic_grid. That's why it fails after loading the default.

noroadsleft, 2020-06-19:
Yeah, apparently something in the API has changed and it no longer picks up the aliases for the Preonic.

Gimme a minute...

noroadsleft, 2020-06-19:
Fixed by 0b035229185fca4d04ad1403c713348d5ce4d427.