qmk/qmk_configurator #755: Support autoshift

GauthierPLM, 2020-06-20:
It would be awesome if autoshift could be enabled from the UI, similarly to Oryx, the ZSA configuration tool for their EZ keyboards.

Having the possibility to select the autoshift control keys would be a good addition too, as now I need to manually customise the firmware to have these.

colelawrence, 2020-11-12:
Can someone give context for how much is already in place, versus how much would need to be added?

Is it just adding button placeholders to the UI? Or is it a combination of adding the JSON action (e.g. `AutoShift[...]`) and the UI creation button?

yanfali, 2020-11-12:
Since it's a rules.mk change, this support would needed to be added to QMK cli first, then it could be supported in the UI.

skullydazed, 2020-11-12:
I have a [branch I've been working on](https://github.com/qmk/qmk_firmware/tree/info_json_config) that adds this (as well as a lot more.) I'm targeting the first develop merge in 2021.

mattrayner, 2022-05-09:
@skullydazed any update on this one?