qmk/qmk_configurator #763: Error in configurator

Crankatator, 2020-06-27:
I'm getting this error:
[image]
 for this keymap:
[image]
The suggested "}" doesn't make sense.  What's going on?

yanfali, 2020-06-27:
Hey are you using cli? File the bug under firmware. But honestly I think you have a typo in your code. It's kc_lctl not kc_lctrl

closed 2020-06-27 by yanfali

noroadsleft, 2020-06-28:
> I'm getting this error:
> [image](https://user-images.githubusercontent.com/67513446/85926156-61cc1e80-b86b-11ea-93d4-cf6eb121612e.png)
> for this keymap: [...]

Line 8 of your `keymap.c`, you're missing a comma after `KC_DEL`.

> I think you have a typo in your code. It's kc_lctl not kc_lctrl

For the record, either should work.

https://github.com/qmk/qmk_firmware/blob/751a1789b59546728c0987d9259b4189cd9f6fe2/tmk_core/common/keycode.h#L153

Crankatator, 2020-06-28:
> image

Thanks for your help!