qmk/qmk_configurator #764: Layout for Helix keyboard won't compile

coaxial, 2020-06-28:
On the configurator, I select the `helix/rev2/oled` keyboard and use this JSON layout. When I press `Compile`, I get the following error.

JSON layout: 

``` json
{"version":1,"notes":"","documentation":"\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n","keyboard":"helix/rev2/oled","keymap":"coaxial","layout":"LAYOUT","layers":[["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_BSPC","KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_BSPC","KC_ESC","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","KC_NO","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_LBRC","KC_RBRC","KC_N","KC_M","KC_COMM","KC_DOT","KC_SLSH","KC_UP","KC_LCTL","TO(10)","TT(11)","KC_LALT","KC_LSFT","KC_ENT","LCTL(KC_A)","OSL(9)","KC_SPC","TT(8)","KC_RSFT","KC_LEFT","KC_RGHT","KC_DOWN"],["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_DEL","KC_TAB","KC_Q","KC_W","KC_F","KC_P","KC_G","KC_J","KC_L","KC_U","KC_Y","KC_SCLN","KC_BSPC","KC_LCTL","KC_A","KC_R","KC_S","KC_T","KC_D","KC_H","KC_N","KC_E","KC_I","KC_O","KC_QUOT","KC_LSFT","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_LBRC","KC_RBRC","KC_K","KC_M","KC_COMM","KC_DOT","KC_SLSH","KC_ENT","MO(5)","KC_TRNS","KC_TRNS","KC_LGUI","KC_NO","MO(3)","KC_SPC","KC_SPC","MO(4)","KC_NO","KC_LEFT","KC_DOWN","KC_UP","KC_RGHT"],["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_BSPC","KC_TAB","KC_QUOT","KC_COMM","KC_DOT","KC_P","KC_Y","KC_F","KC_G","KC_C","KC_R","KC_L","KC_DEL","KC_LCTL","KC_A","KC_O","KC_E","KC_U","KC_I","KC_D","KC_H","KC_T","KC_N","KC_S","KC_SLSH","KC_LSFT","KC_SCLN","KC_Q","KC_J","KC_K","KC_X","KC_LBRC","KC_RBRC","KC_B","KC_M","KC_W","KC_V","KC_Z","KC_ENT","MO(5)","KC_ESC","KC_LALT","KC_LGUI","KC_NO","MO(3)","KC_SPC","KC_SPC","MO(4)","KC_NO","KC_LEFT","KC_DOWN","KC_UP","KC_RGHT"],["KC_TILD","KC_EXLM","KC_AT","KC_HASH","KC_DLR","KC_PERC","KC_CIRC","KC_AMPR","KC_ASTR","KC_LPRN","KC_RPRN","KC_TRNS","KC_TILD","KC_EXLM","KC_AT","KC_HASH","KC_DLR","KC_PERC","KC_CIRC","KC_AMPR","KC_ASTR","KC_LPRN","KC_RPRN","KC_TRNS","KC_TRNS","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_UNDS","KC_PLUS","KC_LCBR","KC_RCBR","KC_PIPE","KC_CAPS","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_LPRN","KC_RPRN","KC_F12","KC_TRNS","KC_TRNS","KC_HOME","KC_END","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","MO(5)","KC_TRNS","KC_MNXT","KC_VOLD","KC_VOLU","KC_MPLY"],["KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_BSPC","KC_GRV","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_DEL","KC_TRNS","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_MINS","KC_EQL","KC_LBRC","KC_RBRC","KC_BSLS","KC_CAPS","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_TRNS","KC_TRNS","KC_F12","KC_TRNS","KC_TRNS","KC_PGDN","KC_PGUP","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","MO(5)","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_MNXT","KC_VOLD","KC_VOLU","KC_MPLY"],["KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_F12","KC_TRNS","RESET","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_DEL","KC_TRNS","KC_TRNS","KC_TRNS","ANY(AU_ON)","ANY(AU_OFF)","AG_NORM","AG_SWAP","TO(0)","TO(1)","TO(2)","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","RGB_TOG","RGB_HUI","RGB_SAI","RGB_VAI","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","KC_TRNS","RGB_MOD","RGB_HUD","RGB_SAD","RGB_VAD"],[],[],["KC_PWR","KC_SLEP","KC_WAKE","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_CUT","KC_COPY","KC_PSTE","KC_NO","KC_NO","KC_CALC","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_LEFT","KC_DOWN","KC_UP","KC_RGHT","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_VOLU","KC_NO","KC_WBAK","KC_WREF","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_VOLD","KC_NO","KC_NO","KC_TRNS","KC_NO","KC_NO","KC_NO","KC_NO"],["KC_NO","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_F7","KC_F8","KC_F9","KC_F10","KC_NO","KC_NO","KC_F11","KC_F12","KC_NO","KC_NO","KC_NO","KC_NO","KC_EQL","KC_MINS","KC_NO","KC_NO","KC_NO","KC_NO","KC_EXLM","KC_AT","KC_HASH","KC_DLR","KC_PERC","KC_CIRC","KC_AMPR","KC_ASTR","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_LPRN","KC_SCLN","KC_COLN","KC_RPRN","KC_DOT","KC_COMM","KC_QUES","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_LSFT","KC_NO","KC_NO","KC_TRNS","KC_BSPC","KC_NO","KC_RSFT","KC_NO","KC_NO","KC_NO"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_MS_L","KC_MS_D","KC_MS_U","KC_MS_R","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_WH_D","KC_WH_U","KC_NO","KC_NO","KC_NO","KC_TRNS","KC_NO","KC_NO","KC_NO","KC_NO","TO(0)","KC_NO","KC_BTN1","KC_BTN2","KC_NO","KC_NO","KC_NO","KC_NO"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_PSLS","KC_PAST","KC_PMNS","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_P7","KC_P8","KC_P9","KC_PPLS","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_P4","KC_P5","KC_P6","KC_PCMM","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_P1","KC_P2","KC_P3","KC_PEQL","KC_NO","KC_NO","KC_NO","KC_TRNS","KC_NO","KC_NO","KC_NO","TO(0)","KC_NO","KC_NO","KC_P0","KC_P0","KC_PDOT","KC_PENT","KC_NO"]],"author":""}
```

```
Helix
===

![Helix](https://i.imgur.com/XBAmynN.jpg)

A compact split ortholinear keyboard.

Keyboard Maintainer: [Makoto Kurauchi](https://github.com/MakotoKurauchi/) [@pluis9](https://twitter.com/pluis9)  
Hardware Supported: Helix PCB Alpha, Beta, Pro Micro  
Hardware Availability: [PCB &amp; Case Data](https://github.com/MakotoKurauchi/helix), [Yushakobo Shop](https://yushakobo.jp/shop/), [Little Keyboards](https://littlekeyboards.com/collections/helix) 

## How to build
 * [Helix how to Customize and Compile](rev2/keymaps/default/readme.md#customize)
 * [HelixPico how to Customize and Compile](pico/keymaps/default/readme.md#customize)

See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.
* Sending helix/rev2/oled:coaxial with LAYOUT
* Received job_id: 8e34ba30-9d72-4439-b4ac-458795124065
* Running . . . . .
* Finished:



Making 

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/helix/helix.c                                                                  
Compiling: keyboards/helix/rev2/rev2.c                                                              
Compiling: keyboards/helix/rev2/keymaps/coaxial/keymap.c                                            
Compiling: quantum/quantum.c                                                                        
Compiling: quantum/keymap_common.c                                                                  
Compiling: quantum/keycode_config.c                                                                 
Compiling: quantum/matrix_common.c                                                                  
Compiling: quantum/matrix.c                                                                         
Compiling: quantum/debounce/sym_g.c                                                                 
Compiling: quantum/process_keycode/process_space_cadet.c                                            
[... cut ...]
Linking: .build/helix_rev2_oled_coaxial.elf                                                         
 | 
 | .build/obj_helix_rev2_oled_coaxial/keyboards/helix/rev2/rev2.o: In function `matrix_init_kb':
 | /qmk_compiler/qmk_firmware/keyboards/helix/rev2/rev2.c:25: undefined reference to `is_helix_master'
 | collect2: error: ld returned 1 exit status
 | 
makek:306: .build/helix_rev2_oled_coaxial.elf] Error 1
make: *** akefile:584: helix/rev2/oled:coaxial] Error 1
ake finished with errors

```

I would expect it to compile without errors, but it doesn't.

noroadsleft, 2020-06-29:
The Helix needs a refactor on the qmk_firmware side in order to fix this. There are a couple of open pull requests attempting to deal with this.

- https://github.com/qmk/qmk_firmware/pull/8422 (deals with the build error)
- https://github.com/qmk/qmk_firmware/pull/8669 (attempting to reduce the number of Helix variants)

mtei, 2020-12-11:
The response in the comment written by noroadsleft above is the best solution to this problem.
However, there is still no agreement, so I have created a temporary workaround.

Currently the following two targets are compilable.
* helix/pico/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)
* helix/rev2/qmk_conf (The setting in rules.mk for the keymap level will be ignored.)

fauxpark, 2022-12-13:
I believe this is fixed now.

closed 2022-12-13 by fauxpark