qmk/qmk_configurator #768: Unable to add ISO-specific keys in configurator

IonicOwl, 2020-06-29:
I'm trying to flesh out an ISO keyboard layout on the configurator, but despite using the keys in the ISO/JIS tab, when I drag them up to the keymap, they show the ANSI key instead. 

In the example below, dragging the KC_2 key up to its position on the top row shows the "2/@" key, instead of the "2/"" key that I'd expect.

[image]

This also occurs with the KC_3 (should be "3/£), KC_GRV and KC_QUOT keys.

Also, since the AltGr key isn't under the ISO tab, am I right in assuming that I should be using the RALT(kc) key under the Quantum tab for accessing graph/accent keys?

yanfali, 2020-06-29:
@IonicOwl this is a good catch, it's more of a visual oversight than an actual bug. Your operating system is responsible for mapping the ANSI based HID codes onto the locale. I think to fix this it mostly requires fixing up the keycodes for the ISO keymap programmatically to display the correct codes. No actual keycodes need to change.

yanfali, 2020-06-29:
Can you send a list of the keys that need updating?

noroadsleft, 2020-06-30:
> Also, since the AltGr key isn't under the ISO tab, am I right in assuming that I should be using the RALT(kc) key under the Quantum tab for accessing graph/accent keys?

Actually, I noticed that the AltGr key was missing from the ISO tab a while back, and I think I may have a branch that adds it.

Gimme some time and I'll dive in to that. The rest of this is as yanfali said – a visual oversight, rather than a bug.

IonicOwl, 2020-06-30:
> Can you send a list of the keys that need updating? or are the ones in the bug exhaustive?

Those are the only ones that need adding for my layout.  I'm only building a 60% and I'm pretty sure everything on the navigation and number clusters are identical between ISO and ANSI layouts anyway.

noroadsleft, 2020-07-01:
#770 will add the Alt Gr key to the ISO tab.