qmk/qmk_configurator #772: Suddenly couldn't export anymore (Firefox 78)

CreamyCookie, 2020-07-03:
Not sure what caused this, but I lost all progress, since out of nowhere the export button stopped working.

I could be wrong but the export function seems to involve a server.. Couldn't it work purely client side since all the information is already available?

When I open FF developer tools console, there are lots of error messages:

https://gist.github.com/CreamyCookie/3ffd92a6ecd1c1616198dcc95273b16a

noroadsleft, 2020-07-05:
I never spotted this, and I use Firefox as my main browser.

Just attempted and it's working for me (Firefox 78.0.1).

@yanfali Any ideas?

yanfali, 2020-07-05:
Try incognito mode. It's probably a plugin

yanfali, 2020-07-19:
@CreamyCookie any progress on this?

CreamyCookie, 2020-07-19:
@yanfali Oh, sorry. I wasn't aware you were referring me. Are you asking about the browser plugins? I'm using uBlock Origin and uMatrix but neither block anything in the qmk configurator page.

I don't remember, but it's possible I opened the configurator in a private window.

yanfali, 2020-07-20:
@CreamyCookie since we can't reproduce this, I'm at a loss. Do you have any other computers? this could be an issue with that one machine. Please try a incognito session to remove all influence of plugins, if you don't respond I'll probably close this.

CreamyCookie, 2020-07-20:
Yeah, this seems to be one of those hard to find bugs. I myself can and could export dozens of times until it suddenly stopped working, and I have no idea what caused it.

If the log is of no help, then I guess we can close this for now.

closed 2020-07-20 by CreamyCookie