qmk/qmk_configurator #785: Add option to duplicate layer

sepehr500, 2020-08-05:
There are times when I want to make a new layer but with slight tweaks. It would be nice if there was an option to duplicate a layer. Does this feature currently exist?

yanfali, 2020-08-05:
Not at this time. Easiest thing in the short term is to edit the JSON file and copy the layer and reimport. I am planning on looking at making the layer editing more featured, just have no time line. Maybe this weekend as I have some free time coming up. Mostly moving layers around, copying would be fairly easy to add to this list.

sepehr500, 2020-08-06:
I can try adding it if you want. Can I just do a reasonable design?

yanfali, 2020-08-06:
> I can try adding it if you want. Can I just do a reasonable design?

Sure, always happy to get input. Design is probably the hardest part.

yanfali, 2020-08-06:
@sepehr500 be aware though that duplicating a layer, leads on automatically to the need to be able to re-order layers 😄

sepehr500, 2020-08-06:
Lol. We can iterate. 😉

sepehr500, 2020-08-06:
Created PR #786

ItamarMM, 2023-11-09:
> Sorry no. I maybe over the holidays I'll look into this again.

You did not :(

morgantic, 2024-01-28:
Any chance for an update on this? I'm loving the flexibility of VIA but without this feature at can be so tedious at times...

plovinicius, 2024-07-09:
is this still a wanted feature? if so, maybe I can try to work on it since I'm looking forward to contribute to the project. Also, any ideas of how the duplicate feature would work? I was thinking of a button to duplicate the selected layer side by side with the `clear layer` button.