qmk/qmk_configurator #788: Where are layouts stored?

iandoug, 2020-08-08:
HI

I need to add a new physical layout, but can't find where they are stored. I've done text searches etc on local copy and here.

Can you please advise? :+1: 

Thanks, Ian

noroadsleft, 2020-08-08:
@iandoug,

QMK Configurator takes its data through an API that runs based upon the current state of https://github.com/qmk/qmk_firmware/tree/master/. If you need to add a physical layout, it needs to be done on that repo.

Which keyboard and layout do you need to add?

iandoug, 2020-08-08:
Okay let me poke around in the firmware repo then, thanks.

Trying to build this (still... :-) )
https://www.keyboard-design.com/letterlayout.html?layout=x6-5h.en.ergolinear

I haven't tried to get the configurator running locally yet ... will that also pull from the net, or look for a local source?

Thanks, Ian

noroadsleft, 2020-08-09:
Ah, a new device.

You'll have to have your codebase merged into qmk/qmk_firmware generally to support a new keyboard. You can also feed it custom API data if you deploy the Configurator in Docker, but I have no experience with that.

iandoug, 2020-08-09:
When I last dipped my toes into QMK a few years back, I vaguely remember some "battleship" layouts that catered for x rows by y columns, where x and y are larger than normal ANSI. Are those still in the configurator?

BTW my mind works the other way around to the configurator... first pick physical layout, and then logical. At the moment once you pick logical, the physical options are restricted, so there is no easy way to browse the physical variants available.

Or maybe I'm missing something obvious ...

Thanks, Ian

yanfali, 2020-08-09:
I think this conversation has been resolved. Closing issue.

closed 2020-08-09 by yanfali