qmk/qmk_configurator #792: Download keymap config button removed?

dainkaplan, 2020-08-14:
There used to be a button to download the keymap config (it's still in the i18n csvs as a label), but it looks to be removed in 
 commit f67c15f92ac52553c67cb9ac5353f31a267d99c0 (via issue #692):

[image]

Downloading the entire source doesn't seem to include the keymap config, so there is now no way to get the config files to modify them.

What I see:
[image]

What the tutorial video shows (and what used to also show previously):
[image]

yanfali, 2020-08-14:
https://github.com/qmk/qmk_configurator/issues/784

yanfali, 2020-08-14:
@skullydazed did you remove the json file from the full source download?

closed 2020-08-14 by yanfali

noroadsleft, 2020-08-16:
> did you remove the json file from the full source download?

JSON file is in the root `qmk_firmware` directory inside the Full Source download.