qmk/qmk_configurator #813: please add support for "U" shaped coulumns.

Supermagnum, 2020-09-10:
A example is this schema, the columns here are wired in "U" shapes to conserve pins.
https://github.com/Supermagnum/98keys-ergo/blob/master/joined-boards.pdf

noroadsleft, 2020-09-10:
QMK Configurator doesn't contain any information about the electrical schematics of a board; all of that is defined in [qmk_firmware](https://github.com/qmk/qmk_firmware/tree/master).

The QMK Docs should hopefully shed some light on this topic:

- [Supporting Your Keyboard in QMK Configurator](https://docs.qmk.fm/#/reference_configurator_support)

Feel free to reply here if you have any questions.

Supermagnum, 2020-09-10:
> QMK Configurator doesn't contain any information about the electrical schematics of a board; all of that is defined in [qmk_firmware](https://github.com/qmk/qmk_firmware/tree/master).
>
> The QMK Docs [...]

The manual doesn't mention anything about a scenario like this, I will file a issue on it and hopefully it will be fixed.

noroadsleft, 2020-09-11:
The `clueboard/66/rev4` does something like this - the columns aren't U-shaped, but more like a backwards N (edited for clarity/readability):

```c
#define LAYOUT_66_ansi( \
    k00, k01, k02, k03, k04, k05, k06, k07, k50, k51, k52, k53, k54,   k56,    k57, \
    k10, k11, k12, k13, k14, k15, k16, k17, k60, k61, k62, k63, k64, k65,      k67, \
    k20, k21, k22, k23, k24, k25, k26, k27, k70, k71, k72, k73,    k75, \
    k30,      k32, k33, k34, k35, k36, k37, k80, k81, k82, k83,    k85,   k86, \
    k40, k41, k42,                k46,                k92, k93, k94, k95, k96, k97 \
) { \
    { k00, k01, k02, k03, k04, k05, k06, k07 }, \
    { k10, k11, k12, k13, k14, k15, k16, k17 }, \
    { k20, k21, k22, k23, k24, k25, k26, k27 }, \
    { k30, XXX, k32, k33, k34, k35, k36, k37 }, \
    { k40, k41, k42, XXX, XXX, XXX, k46, XXX }, \
    { k50, k51, k52, k53, k54, XXX, k56, k57 }, \
    { k60, k61, k62, k63, k64, k65, XXX, k67 }, \
    { k70, k71, k72, k73, XXX, k75, XXX, XXX }, \
    { k80, k81, k82, k83, XXX, k85, k86, XXX }, \
    { XXX, XXX, k92, k93, k94, k95, k96, k97 }  \
}
```

The keys for 8, I, K and M (`k50`, `k60`, `k70` and `k80`) are on the same column pin as those for Escape, Tab, Caps Lock, Left Shift and Left Control (`k00`, `k10`, `k20`, `k30` and `k40`).

- https://github.com/qmk/qmk_firmware/blob/3d4f0028d60cebc829ac9c947d1a61cc840d05c0/keyboards/clueboard/66/rev4/rev4.h#L53-L70

closed 2020-09-14 by nooges