qmk/qmk_configurator #816: kbd67mkii ISO ~ character not mapped

Mouzourides, 2020-09-19:
**PCB**: KBD67mkii soldered 
**Selected keyboard**: kbdfans/kbd67/mkii_soldered
**Layout**: LAYOUT_65_iso_blocker

**Problem**: The key left of ISO enter is not getting mapped correctly.

**Description**:
When selecting the above parameters, creating the HEX file and using QMK Toolbox to flash the firmware, the ~ key, left of the ISO Enter does not get mapped at all. Pressing the key results in nothing being input and keyboard tester not registering an input.

I have tried this on two different KBD67mkii soldered PCBs and get the same result. When flashing an ANSI layout, the key works as an Enter key, suggesting that the key is not being mapped correctly specifially for ISO layout.

noroadsleft, 2020-09-19:
@Mouzourides,

I don't think this is going to be a QMK Configurator problem directly, but could you flash this HEX file to your board and report back to me?

- https://raw.githubusercontent.com/noroadsleft/qmk_configurator/fa7b072445723f03dfb2494d62dd9889c4c9a14f/public/keymaps/k/kbdfans_kbd67_mkii_soldered_iso.hex

It's [the ISO keymap for the board](https://github.com/qmk/qmk_firmware/blob/a0efe932394492ddd79212c2e652306aae9cf72d/keyboards/kbdfans/kbd67/mkii_soldered/keymaps/iso/keymap.c), but I modified the matrix data and compiled it locally.

Mouzourides, 2020-09-19:
Hi @noroadsleft, thanks for the quick response.
Flashing that HEX seems to result in the key in question, (KC_NUHS by default on ISO) key being mapped as an Enter key.

noroadsleft, 2020-09-19:
What happens if you hit the physical Enter key?

Mouzourides, 2020-09-19:
That seems to now be mapped to KC_NUHS (#~)

noroadsleft, 2020-09-19:
K, try this (right-click; Save Link As...):

- https://raw.githubusercontent.com/noroadsleft/qmk_configurator/c19e4d54d6338973a990a79e1c85fbb9cddcdce9/public/keymaps/k/kbdfans_kbd67_mkii_soldered_iso.hex

Mouzourides, 2020-09-19:
That works :) KC_NUHS and Enter key work as expected

noroadsleft, 2020-09-19:
Excellent!

This needs to be fixed on `qmk_firmware` repository. I'll make the PR.

closed 2020-09-21 by Mouzourides

noroadsleft, 2020-09-21:
@Mouzourides Thanks for reporting this, by the way.

Mouzourides, 2020-09-21:
Thanks for fixing it so quickly :smile: