qmk/qmk_configurator #826: Keymap via URL not working

bronzehedwick, 2020-10-03:
Not sure if this is an isolated incident, but trying to reference a keymap via this URL results in a non-specific error window: https://www.chrisdeluca.me/page/keyboard/keymap.json

Uploading the same JSON as a file works fine. I double checked the file encoding on the server side and it's `application/json`, so not sure what could be going wrong, and without a more verbose error message having trouble narrowing it down.

Thanks!

noroadsleft, 2020-10-04:
Attempted to replicate this and got the following in my JavaScript console:

```
16:00:11.460 Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://www.chrisdeluca.me/page/keyboard/keymap.json. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing).
```

This may only work when fed GitHub-hosted URLs (QMK Configurator is hosted through GitHub Pages). I don't normally use this feature, but last I checked it it was working, and I can't think of any changes that would've broken it.

yanfali, 2020-10-05:
You need to set CORS headers that permit cross origin json requests.

bronzehedwick, 2020-10-05:
Ahh yes, that makes sense, thanks.

closed 2020-10-05 by bronzehedwick