qmk/qmk_configurator #831: New custom hand-wired keyboard

BB-66, 2020-10-17:
Greetings. I am working on a personal project, it's a custom keyboard that will be hand-wired and will use an elite-c controller. At the moment I've devised the matrix to be 9x12 as attached and will have the layout attached - it will use an Elite-C v3.1 so it's why I had to get creative with the rows and columns.

[image]
[image]

[This](https://imgur.com/a/TvsXpq5) is the album with the hardware work underway.

How do I get to use qmk configurator with this? When I upload the JSON it doesn't allow it. I get I could do a pull from QMK via Visual Studio Code and create a new branch? I apologise, I'm new to github, programming and all this.

Thank you :)

noroadsleft, 2020-10-17:
There's [a page in the QMK Docs](https://docs.qmk.fm/#/reference_configurator_support) on how to do this.

The short of it is that QMK Configurator only compiles for keyboards that are present in the [`qmk/qmk_firmware` repository](https://github.com/qmk/qmk_firmware/tree/master/keyboards). You'll want to fork that repo, create a new branch, add your source files to that branch, and open a pull request from that branch to the `master` branch on https://github.com/qmk/qmk_firmware/. Your source files should go in the `keyboards/handwired/` directory.

If you need any more guidance let me know, or hop on the [QMK Discord](https://discord.gg/Uq7gcHh).

BB-66, 2020-10-17:
@noroadsleft thank you so much for your response. I have setup my environment on all my computers and already have made some strides into porting some of my keyboards to VIA - unsuccessful at first but I learned a lot. But the whole github environment part is a whole new thingie, I've worked only off local files prior to this.

Will read on forking the repo, creating a new branch and creating the source files plus the pull request and will crack on to work then. Thank you again!

Also as a side note the QMK discord is incredibly helpful, they managed to unfreeze my brain on how to stick the whole above layout on only 21 pins of the controller ^^

noroadsleft, 2020-10-17:
> Will read on forking the repo, creating a new branch and creating the source files plus the pull request and will crack on to work then. Thank you again!

This is a good quick guide on dealing with forks: https://github.com/k88hudson/git-flight-rules#repositories

Be advised that the method given there for "I need to update my fork with latest updates from the original repository" is actually **not recommended** by QMK, and we suggest an alternative method, detailed here: https://docs.qmk.fm/#/newbs_git_best_practices

BB-66, 2021-04-09:
The build is now [complete](https://imgur.com/a/TvsXpq5).

[Typing test](https://vimeo.com/534299400).

@noroadsleft expect poking soon for how to add to QMK :D I did the forking, but having trouble creating the branch, highly likely this is just me being inexperienced with github.

noroadsleft, 2021-04-11:
@BB-66,

Go to your `qmk_firmware` fork, and look for this button and click it:

[image]

That will get you a drop down menu like this:

[image]

In the "Find or create a branch" input field, type your desired name for the new branch, and click "Create branch: <typed name> from master":

[image]

That will create your new branch and switch to it. From here you can add your files (there will be a button for it across from where your current branch is shown), or edit files that are already present.

BB-66, 2021-04-11:
I have created the pull request now, I hope I didn't mess anything up on the way. Thank you so much for your help!

fauxpark, 2022-12-13:
Above PR is merged, assuming this issue is resolved.

closed 2022-12-13 by fauxpark