qmk/qmk_configurator #838: 'Unknow error' ISO KBD67 MKii soldered

FruitLoopsGB, 2020-11-06:
Trying to compile and an unknown error occurs every time.
[
[image]
](url)

closed 2020-11-06 by FruitLoopsGB

noroadsleft, 2020-11-08:
We had some issues this past week with the infrastructure on which the Configurator runs – you may have been trying to compile during that time. I tried compiling this board and can't reproduce this issue now.

Sorry for the troubles, either way.

ostap-tymchenko, 2023-11-28:
Hi! im getting this issue for 2 days now, and i cant compile anything. It started late last night/ early morning est.
[image]

keyboard-magpie, 2023-11-28:
Much the same as per noroadsleft's message on Nov 8 2020, we've just had a breaking changes cycle so some things require updating. The team are [working on it](https://github.com/qmk/qmk_firmware/pull/22555).