qmk/qmk_configurator #840: KBD19X Missing Objects for VIA Design

bvbowes, 2020-11-09:
Exported a JSON file from online QMK and when imported to JSON in VIA its missing the following:
Object: should have required property 'layouts'
Object: should have required property 'lighting'
Object: should have required property 'matrix'
Object: should have required property 'productid'
Object: should have required property 'vendorId'

yanfali, 2020-11-09:
Sorry these projects are not compatible. Via has their own proprietary format. You'll need to check their docs for details.

closed 2020-11-09 by yanfali

noroadsleft, 2020-11-10:
@bvbowes You need to follow the VIA Docs: https://caniusevia.com/docs/specification

VIA JSON files go here: https://github.com/the-via/keyboards/tree/master/src