qmk/qmk_configurator #846: [Feature Request] Support for other software keyboard layouts

Ingvix, 2020-11-18:
I think it'd be nice to be able so see the keys represented as they are in the non-ANSI keyboard layout I use on my devices.

Someone hinted that Freedesktop group maintains a pretty comprehensive database of the different keyboard layouts and theoretically it could be queried to provide the functionality to select according to which layout should the keycodes be visually represented in QMK Configurator.

yanfali, 2020-11-18:
This is a surprisingly complicated task. Basically all keyboards, at the firmware level are actually ANSI over HID USB. The OS is actually responsible for translating those key positions into the locale for you, so we could take the translation header files in theory and provide a mapping, but it's really a function of your operating system, not the keyboard firmware as what the meaning of each of those keycodes is.

Ingvix, 2020-11-18:
I understand and it's not that huge problem to just deal with it. It would just be a nice visual help when using the configurator to see the keys as you've learned to know them rather than checking all the time what was this and that key's equivalent in my local layout. Until I've done it long enough to memorize the standard ANSI layout entirely.