qmk/qmk_configurator #873: make: *** ain.o] Error 1

mvksym, 2021-01-20:
Hi, I'm trying to flash my NK65 and trying to compile using config.qmk.fm and whenever I do I get,

make: *** ain.o] Error 1

at the bottom and am unsure on what to do - I erased all layers and redid them to make sure there was no overlapping keys or any weird keys - I also used a unique name.

Any help would be appreciated.

noroadsleft, 2021-01-20:
The full output from my attempt to compile the included default keymap:

```
Ψ Compiling keymap with make -s -j 1 -r -R -f build_keyboard.mk GIT_VERSION=4e8166-dirty BUILD_DATE=2021-01-20-19:55:12 CHIBIOS_VERSION=breaking_2020_q1 CHIBIOS_CONTRIB_VERSION=breaking_2020_q1 KEYBOARD=nk65 KEYMAP=layout_65_ansi_mine KEYBOARD_FILESAFE=nk65 TARGET=nk65_layout_65_ansi_mine KEYBOARD_OUTPUT=.build/obj_nk65 KEYMAP_OUTPUT=.build/obj_nk65_layout_65_ansi_mine MAIN_KEYMAP_PATH_1=.build/obj_nk65_layout_65_ansi_mine MAIN_KEYMAP_PATH_2=.build/obj_nk65_layout_65_ansi_mine MAIN_KEYMAP_PATH_3=.build/obj_nk65_layout_65_ansi_mine MAIN_KEYMAP_PATH_4=.build/obj_nk65_layout_65_ansi_mine MAIN_KEYMAP_PATH_5=.build/obj_nk65_layout_65_ansi_mine KEYMAP_C=.build/obj_nk65_layout_65_ansi_mine/src/keymap.c KEYMAP_PATH=.build/obj_nk65_layout_65_ansi_mine/src VERBOSE=false COLOR=true SILENT=false

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/wilba_tech/wt_main.c                                                          keyboards/wilba_tech/wt_main.c: In function 'via_eeprom_is_valid':
keyboards/wilba_tech/wt_main.c:169:15: error: 'QMK_BUILDDATE' undeclared (first use in this function)
     char *p = QMK_BUILDDATE; // e.g. "2019-11-05-11:29:54"
               ^~~~~~~~~~~~~
keyboards/wilba_tech/wt_main.c:169:15: note: each undeclared identifier is reported only once for each function it appears in
keyboards/wilba_tech/wt_main.c: In function 'via_eeprom_set_valid':
keyboards/wilba_tech/wt_main.c:181:15: error: 'QMK_BUILDDATE' undeclared (first use in this function)
     char *p = QMK_BUILDDATE; // e.g. "2019-11-05-11:29:54"
               ^~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make: *** [tmk_core/rules.mk:412: .build/obj_nk65_layout_65_ansi_mine/keyboards/wilba_tech/wt_main.o] Error 1
```

mvksym, 2021-01-20:
In response to the comment, I should've included this.. This is what shows on config.qmk.fm

NK65
=========

[image]

This is a standard fixed layout 65% PCB. It supports VIA and full per-key RGB.

Keyboard Maintainer: [Yiancar](http://yiancar-designs.com/) and on [GitHub](https://github.com/yiancar)   
Hardware Supported: A 65% keyboard with STM32F303CC   
Hardware Availability: https://novelkeys.xyz/   

Due to the RGB implementation, the NK65 is currently not compatible with community layouts.

## Instructions

### Build

Make example for this keyboard (after setting up your build environment):

    make nk65:default_via

See the [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools) and the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information. Brand new to QMK? Start with our [Complete Newbs Guide](https://docs.qmk.fm/#/newbs).

### Reset

- Unplug
- Hold Escape
- Plug In
- Unplug
- Release Escape

### Flash

- Unplug
- Hold Escape
- Plug In
- Flash using QMK Toolbox or dfu-util (`make nk65:&lt;keymap&gt;:dfu-util`)

Loaded 4 layers and 272 keycodes. Defined 14 Any key keycodes
* Sending nk65:mk_nk65 with LAYOUT_65_ansi
* Received job_id: 36456f9e-4921-43ef-a1ed-b119a825a686
* Running . . . . . . . . . . . .
* Finished:


arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) bedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/wilba_tech/wt_main.c                                                          keyboards/wilba_tech/wt_main.c: In function 'via_eeprom_is_valid':
keyboards/wilba_tech/wt_main.c:169:15: error: 'QMK_BUILDDATE' undeclared (first use in this function)
     char *p = QMK_BUILDDATE; // e.g. "2019-11-05-11:29:54"
               ^~~~~~~~~~~~~
keyboards/wilba_tech/wt_main.c:169:15: note: each undeclared identifier is reported only once for each function it appears in
keyboards/wilba_tech/wt_main.c: In function 'via_eeprom_set_valid':
keyboards/wilba_tech/wt_main.c:181:15: error: 'QMK_BUILDDATE' undeclared (first use in this function)
     char *p = QMK_BUILDDATE; // e.g. "2019-11-05-11:29:54"
               ^~~~~~~~~~~~~
 
 | 
 | 
 | 
make: *** ain.o] Error 1

noroadsleft, 2021-01-20:
Seems there's an issue with the generation of `quantum/version.h` here. I downloaded the Full Source from the above compile attempt, and the required file is present, but the syntax is wrong

`quantum/version.h` from the above compile:

```
#define QMK_VERSION "4e8166-dirty"
#define QMK_BUILD_DATE "2021-01-20-19:55:12"
#define CHIBIOS_VERSION "breaking_2020_q1"
#define CHIBIOS_CONTRIB_VERSION "breaking_2020_q1"
```

`quantum/version.h` from running `make nk65:default` on a cloned repo:

```
#define QMK_VERSION "0.11.53-7-g4e8166"
#define QMK_BUILDDATE "2021-01-20-12:12:29"
#define CHIBIOS_VERSION "breaking_2020_q1"
#define CHIBIOS_CONTRIB_VERSION "breaking_2020_q1"
```

The correct definition is `QMK_BUILDDATE`, but the Configurator compile generates `QMK_BUILD_DATE`.

closed 2021-01-20 by zvecr

noroadsleft, 2021-01-20:
@mvksym Just tried this again and it's working:

```
* Sending nk65:layout_65_ansi_mine with LAYOUT_65_ansi
* Received job_id: 440bcc58-2498-4851-b6ed-fb597bfff3d0
* Running . . . . . . . . . . . . . . . . . . . . .
* Finished:


arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) bedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

...

Linking: .build/nk65_layout_65_ansi_mine.elf                                                        
Creating binary load file for flashing: .build/nk65_layout_65_ansi_mine.bin                         
Creating load file for flashing: .build/nk65_layout_65_ansi_mine.hex                                

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  30860	      0	  30860	   788c	.build/nk65_layout_65_ansi_mine.hex

Copying nk65_layout_65_ansi_mine.bin to qmk_firmware folder                                         
(Firmware size check does not yet support cortex-m4 microprocessors; skipping.)
```