qmk/qmk_configurator #876: Split keyboard slave gives no Input [Bug]

flaschen, 2021-01-31:
First time poster here, sorry for any inconvenience. 
So I've successfully flashed new firmwares on this keyboard before. But as of now the newly compiled hex files somehow fail to let the slave part give input.

Exceptions: 

- when I pull the data cable at the slave, sometimes the last typed characters suddenly appear
- when an older hex file (like from April 2020) from my archive is flashed, it works flawlessly (but if I compile a new hex file on basis of the then-used json file, the slave doesn't work)

what I tried:

- double and triple checking the settings of QMK Configurator and Toolbox
- rebooting of course
- cleaning the jacks
- using one of the default layouts of QMK Configurator
- comparing the logs of QMK Toolbox (working vs non-working hex file)

It just has to be a bug.

yanfali, 2021-01-31:
Did you flash both halves? You can't update just one side if it's been a long time between flashing.

flaschen, 2021-02-02:
Wow, thank you! I'd never have thought of this.

closed 2021-02-02 by flaschen