qmk/qmk_configurator #890: Quantum Ralt actuates ctrl [Bug]

Gavus, 2021-02-28:
For every key where I'm using RAlt(key) it also execute left ctrl.

Tested on kinesis stapelberg, flashing with qmk toolbox.

noroadsleft, 2021-03-02:
This sounds to me like it's more an "issue" (in quotes because I'm not convinced you're actually experiencing a bug) with the keyboard layout you have set in your operating system.

For legacy reasons, the Right Alt key is actually frequently treated by the operating system as Ctrl+Alt. You can see this borne out in some keyboard testers by loading the tester, tapping Right Alt, then changing the operating system's layout to US English QWERTY, then tapping Right Alt again. The first layout will fire Ctrl+Alt, but the US English QWERTY will only fire Right Alt.

As your GitHub profile says you are in Sweden, I imagine that's what's happening. If you set your OS to US English QWERTY, I bet it stops firing the Ctrl key.

Reference: [AltGr key (Control + Alt as a substitute) - Wikipedia](https://en.wikipedia.org/wiki/AltGr_key#Control_+_Alt_as_a_substitute)

Gavus, 2021-03-02:
You are completely right, tried US English QWERTY and it did as you said.

closed 2021-03-02 by Gavus