qmk/qmk_configurator #892: Bug - Anything compiled recently will give Redox/rev1 "7676" key stroke outputs

djshorty, 2021-03-01:
Difficult to explain in one sentence, but anything I have compiled recently and installed using the QMK Toolbox "breaks" my keyboard. When I type it will randomly output strings of 767676. If I install it with older .hex files I have saved it is fine. If I compile the same hex within the qmk configurator I have the same issue. (Meaning it is breaking old formats I know work fine) I have tried this over a 3 days span. 

My keyboard is the Redox, and I compile under Redox/rev1. 
I have not tried any other formats as I don't have other QMK boards.

yanfali, 2021-03-01:
Did you flash both halves? There's no guarantee that a split keyboard will work if you only flash one half after a few months of changes in QMK.

djshorty, 2021-03-01:
It appears that did work. I guess I never attempted to flash the other side, and assumed doing it to the master side would handle it. I guess I was wrong. I am sorry for using up your time. 

Thank you for your time <3

closed 2021-03-01 by djshorty

yanfali, 2021-03-01:
You can flash only the master side if both sides have recent images if you're making quick changes. Otherwise it's safer to update both sides if it's been a while.

djshorty, 2021-03-01:
Good to know, thank you!