qmk/qmk_configurator #897: [Bug] Configurator error during compilation

mattrighetti, 2021-03-07:
Compiler encounters error when I try to build my whitefox configuration, setup is following:
```
keyboard: whitefox
layout: LAYOUT_65_ansi_blocker_split_bs
```

## Description
This is my JSON configuration:
```json
{"version":1,"notes":"","documentation":"\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n","keyboard":"whitefox","keymap":"whitefox_mattrighetti","layout":"LAYOUT_65_ansi_blocker_split_bs","layers":[["KC_ESC","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_MINS","KC_EQL","KC_GRV","KC_BSLS","KC_NO","KC_TAB","KC_Q","KC_W","KC_E","KC_R","KC_T","KC_Y","KC_U","KC_I","KC_O","KC_P","KC_LBRC","KC_RBRC","KC_BSPC","KC_DEL","KC_CAPS","KC_A","KC_S","KC_D","KC_F","KC_G","KC_H","KC_J","KC_K","KC_L","KC_SCLN","KC_QUOT","KC_ENT","KC_PGUP","KC_LSFT","KC_Z","KC_X","KC_C","KC_V","KC_B","KC_N","KC_M","KC_COMM","KC_DOT","KC_SLSH","KC_RSFT","KC_UP","KC_PGDN","KC_LCTL","MO(2)","KC_LALT","KC_SPC","KC_RALT","MO(1)","KC_LEFT","KC_DOWN","KC_RGHT"],["KC_NO","KC_F1","KC_F2","KC_F3","KC_F4","KC_F5","KC_F6","KC_F7","KC_F8","KC_F9","KC_F10","KC_F11","KC_F12","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_UP","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_LEFT","KC_DOWN","KC_RGHT","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_VOLU","KC_VOLD","KC_MUTE","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_TRNS","KC_NO","KC_NO","KC_NO"],["KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_1","KC_2","KC_3","KC_4","KC_5","KC_6","KC_7","KC_8","KC_9","KC_0","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_TRNS","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO","KC_NO"]],"author":""}
```
The configuration works fine until I need to compile the firmware, the configurator during compilation quits with the following error:
```
Loaded 3 layers and 204 keycodes. Defined 0 Any key keycodes
* Sending whitefox:whitefox_mattrighetti with LAYOUT_65_ansi_blocker_split_bs
* Received job_id: ef675339-9ca8-4d5a-b9cd-12beb9af00e6
* Queueing . . . .
* Running . . . . . . . .
* Finished:


quantum/visualizer/visualizer.mk:79: lib/ugfx/gfx.mk: No such file or directory
make: *** No rule to make target 'lib/ugfx/gfx.mk'.  Stop.
```

And therefore it won't let me download the firmware

yanfali, 2021-03-08:
Sorry this is a known problem. There's a licensing issue with the ugfx library that input club used and it isn't GPL so we can't distribute it with binary compilation. You will have to download the QMK source and compile it locally.

fauxpark, 2022-12-13:
uGFX is gone, this board builds now.

closed 2022-12-13 by fauxpark