qmk/qmk_configurator #902: [Bug]

freeo, 2021-03-17:
## Describe the Bug

The Ergodash UI layout isn't working as of recently, which makes the configurator completely unusable for this keyboard. See the screenshot. It was fine ~1.5 months ago.
Tested on:
Mac: Chrome
Linux: Chrome,Firefox

Same result everywhere.

## Additional Context

[image]

yanfali, 2021-03-17:
It's missing an info.json on the qmk_firmware side, so the UI can't figure out how to render it. Someone needs to add a valid one.

yanfali, 2021-03-17:
Looks like someone removed the LAYOUT that's default for this keyboard, so it's no longer valid. The API does it's best.

yanfali, 2021-03-17:
Ah ok. This is related to LAYOUT aliasing issue #895 I guess we need to add this soon.

noroadsleft, 2021-03-23:
Can confirm this is fixed:

[image]

closed 2021-03-23 by noroadsleft