qmk/qmk_configurator #907: [Bug]

CGMossa, 2021-04-13:
I think the layout of ortho75 has too many buttons.

[image]

Can someone see if this can be fixed easily?

See this particular layout [here](https://config.qmk.fm/#/cannonkeys/ortho75/LAYOUT_ortho_5x15).

yanfali, 2021-04-13:
This isn't a configurator bug per-se. The info.json in QMK configurator is missing a 5x15 layout.

yanfali, 2021-04-13:
@noroadsleft @awkannan someone needs to update info.json or just use via.

closed 2021-04-13 by yanfali

noroadsleft, 2021-04-14:
This is fixed.

[image]

CGMossa, 2021-04-14:
Thanks! This was very nice of you all!