qmk/qmk_configurator #908: [Bug] cannot compile

swizzley, 2021-04-15:
## Describe the Bug

compilation fails for EVERYTHING

## Additional Context

Loaded 2 layers and 170 keycodes. Defined 0 Any key keycodes
* Sending tkc/m0lly:layout_all_mine with LAYOUT_all
* Received job_id: b8159aa4-036e-4fb6-9db9-40bc92aaeb04
* Queueing .
* Running . . . .
* Finished:
Traceback (most recent call last):
  File "bin/qmk", line 98, in <module>
    main()
  File "bin/qmk", line 79, in main
    import qmk.cli  # noqa
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/cli/__init__.py", line 9, in <module>
    from . import c2json
  File "/qmk_compiler/qmk_firmware/lib/python/qmk/cli/c2json.py", line 20, in <module>
    @cli.subcommand('Creates a keymap.json from a keymap.c file.')
  File "/usr/local/lib/python3.7/dist-packages/milc/milc.py", line 284, in argument_function
    self.subcommandsent(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/milc/configuration.py", line 74, in add_argument
    self.subparser.add_argument(*args, **kwargs)
  File "/usr/lib/python3.7/argparse.py", line 1362, in add_argument
    action = action_class(**kwargs)
TypeError: __init__() got an unexpected keyword argument 'completer'

skullydazed, 2021-04-15:
We just merged a PR that depends on MILC 1.3.0 but the infrastructure is running an older version. I've triggered a build and it should be working again soon.

skullydazed, 2021-04-15:
This should be working again. Thanks for the report!

swizzley, 2021-04-15:
appears good now. thanks for the quick turn around!

swizzley, 2021-04-15:
Just flashed DZ60, but appears to be borked, the PCB is non responsive afterwards, it can RESET and be re flashed but I no longer have a good firmware.

swizzley, 2021-04-15:
Okay I don't know if this is related or not because I tried an old DZ60 hex afterwards with the same result, but the PCB was lit up and functional before I flashed it with the latest baked firmware, and ever since then no lights, no key inputs work at all. PCB is supposed to be VIA compatible and it does not recognize it at all. I can hit the RESET and QMK recognizes it, and flashes successfully every time but the PCB is non responsive.

swizzley, 2021-04-15:
I think I just used the wrong dz60 firmware. I needed the RGB WKL v2.

closed 2021-04-15 by swizzley