qmk/qmk_configurator #911: Proxy being denied access when clicking "Full Source" link [Bug]

tylwills, 2021-04-20:
Attempted multiple times when importing my .jsn file, no issue compiling either, but when I go to my download link it denies me access.

I live in the UK if thats an issue that I'm not aware of. (If I were to be needing to connect to a proxy first or something)

Thanks!

Here's what I can't access:
https://nyc3.digitaloceanspaces.com/qmk-api/da0dda8f-b549-434a-8a6d-a4d6ccacad87/qmk_firmware-dztech-dz60rgb-v2-tylers_60_board.zip?AWSAccessKeyId=GRYDYQMNSYP4OK3WPXBK&Signature=l718y54P6DeBUwM591NNzkYMxj8%3D&Expires=1618962715

yanfali, 2021-04-21:
It's probably a temporary infrastructure issue, @skullydazed is travelling right now so it make take a little while for him to get to this, and it maybe have cleared up on it's own

tylwills, 2021-04-21:
I have solved the issue! I used my mobile data hotspot to connect to withb my PC, page loaded fine and the filfe downloaded successfully. I'll leave this here just in case someone goes through the same struggle- but feel free to delete.

closed 2021-04-21 by tylwills

yanfali, 2021-04-21:
> I have solved the issue! I used my mobile data hotspot to connect to withb my PC, page loaded fine and the filfe downloaded successfully. I'll leave this here just in case someone goes through the same struggle- but feel free to delete.

I think you should probably bring this up with your ISP. It seems they are blocking things they shouldn't, or your DNS isn't resolving things it should. Glad you figured it out