qmk/qmk_configurator #913: Add a kinesis advantage layout to converter/usb_usb/hasu

wwalker, 2021-04-22:
## Description

I'd love to see a Kinesis Advantage shaped layout on the configurator for converter/usb_usb/hasu

These setups all have the advantage shaped layouts:

kinesis/alvicstep
kinesis/kint2pp
kinesis/stapelberg
kinesis/nguyenvietyen

For the life of me I can see where the shape is tied to the keyboards.
I'll do the work, just need a couple of pointers.

yanfali, 2021-04-22:
check info.json

wwalker, 2021-04-22:
Thank you!  I didn't think about it being outside of this repo.

wwalker, 2021-04-23:
How do I get my test configurator to see the changes I make in my local qmk_firmware repo?

I tried changing https://github.com/qmk/qmk_firmware to file:///home/wwalker/qmk_firmware, I assumed that was a security problem, so I changed it to a URL for a locally running webserver.  Same results.

Please what am I missing?  Or do I need to fork `qmk_firmware` and change the URL to point to my fork and push changes between tests?

yanfali, 2021-04-23:
I believe the docker container is the best approach.

wwalker, 2021-04-23:
The docker container is still not looking at my local qwk/qwk_**_firmware_** where I made the changes that should enable kinesis keyboard layout ( adding the layouts for kinesis to `qmk_firmware/keyboards/c/converter/usb_usb/hasu/info.json` )

There is absolutely nothing that I can find that tells qmk_configurator where my qmk_firmware is to get the information from said `info.json`

yanfali, 2021-04-24:
@skullydazed probably has the magical incantation to make it all work

owbear, 2021-08-07:
Did you manage to construct a `LAYOUT_pretty()` you'd Iike to share @wwalker ?

wwalker, 2021-08-08:
@owbear yes, and no.  Below is my map.  I think it is good, but I cannot figure out how to test it.  I can set up a test environemnt, but it is hardcoded somewhere in the javascript guts to talk to some API on the real live site.  Here is the file I created:

https://github.com/wwalker/qmk_firmware/commit/537d0ddea78e7ead1408c6b59d5ce5cf036c090e

owbear, 2021-08-10:
Thanks, I'll check it out when I'm back from my vacation next week