qmk/qmk_configurator #915: [Feature Request] Macros Option

HunterArmstrong, 2021-05-02:
About a month ago I watched a custom keyboard video where the keyboard being made was designed as an emoji keyboard for discord. Long story short, the keyboard was programed via QMK with macros to type these emojis or custom text or phrases. 

After using the QMK configurator. I found that you cant actually do that and have to use a different program to integrate macros into the keyboard; which is a learning curve compared to the online configurator. I think that, as a way to add more functions to a keymap, there could be a macro or "custom phrase" keycode that the user could place then customize the phrase.

For this i was thinking that the KC could be placed in maybe the app, media, and mouse group. and with the keycode there would be a text box where the user could click on it and write the phrase they want to use the key for.

I don't know that much about how the program works or how it functions. So, if there are any issues with this suggestion, please let me know. Otherwise, I hope this is a valid enough suggestion to be approved and integrated. :)

noroadsleft, 2021-05-04:
Macros are definitely on the wishlist, but the file format Configurator uses and the API that glues it all together both need updates before this can happen.

codebymiles, 2021-06-17:
@noroadsleft is there an issue to track the changes needed to support macros?  I'm happy to get involved since I also want macros to be supported in Configurator.

skullydazed, 2021-06-17:
@codebymiles The overall [QMK roadmap is kept here](https://hackmd.io/fpPylbKgQROie6ndKHimfA), and basically what we lack here is a design. Someone needs to do the work to define how custom keycodes work, get buy-in on that, implement it, get it merged, and then do the same for macros. Being a volunteer led and driven project that happens when someone has time to work on that.

I will get to these tasks... eventually. :) If you'd like to work on them sooner I'd encourage you to start by familiarizing yourself with our JSON format (we have [json schemas in the qmk_firmware repository](https://github.com/qmk/qmk_firmware/tree/master/data/schemas)) and to consider how custom keycodes can be defined there and exposed to keymaps.

When you have the start of a design [open an issue in qmk_firmware](https://github.com/qmk/qmk_firmware/issues/new?assignees=&labels=help+wanted%2C+question&template=zzz_blank.md&title=) and tag me and we can work from there.

Weshnaw, 2022-04-07:
Macros are in usable with qmk compile json, and it seems to be working superbly on my planck keyboard with a manually edited json.

Reference:
https://github.com/qmk/qmk_firmware/blob/master/docs/feature_macros.md